ipbus/ipbus-firmware #10: Incorrect checksum in IP header of certain control response packets

tswilliams, 2017-04-10:
The incorrect checksum appears to be calculated for the IPv4 header of some large IPbus control response packets that should have checksum value of 0xfeff. Instead, the firmware calculates a value of 0xffff, due to an issue with a carry bit in the final checksum calculation for a particular corner of the packet length and checksum parameter space

Symptoms:
 * Since the checksum in the IPv4 header checksum is incorrect, the IPbus response packet will be dropped by the network interface on the computer that's running the ControlHub / client application. Therefore this problem will be seen by the control software as repeated packet loss for a particular set of transactions, after a period of successful communication with the hardware.
 * However, these dropped packets can be spotted by running wireshark on the computer. Wireshark will highlight these packets as having an incorrect checksum value in the IP header
 * Since the source IP address is an input to the checksum calculation, the problem will only occur if the board is using certain IP addresses.

tswilliams, 2017-04-11:
Fix - provided by Dave Sankey - is to change the following lines in process `udp_build_data` from `udp_tx_mux.vhd` (line numbers 365 to 373 for `ipbus2_v1` tag) from:
```
        when 11 =>
-- capture cksum...
          ip_cksum_int(15 downto 8) := not outbyte;
        when 12 =>
          ip_cksum_int(7 downto 0) := not outbyte;
-- now start on IP length, first headers 20 + 8 + 4...
          int_data_int := (Others => '0');
        when 13 =>
          int_data_int := x"20"; 
```
... to:
```
        when 12 =>
-- capture cksum...
          ip_cksum_int(7 downto 0) := not outbyte;
-- now start on IP length, first headers 20 + 8 + 4...
          int_data_int := (Others => '0');
        when 13 =>
          ip_cksum_int(15 downto 8) := not outbyte;
          int_data_int := x"20”; 
```

Jeroen confirmed it solved the solve problem encountered in the CMS pixel FEDs.

alessandrothea, 2017-05-20:
This issue seems ready to be closed. Shall we?

tswilliams, 2017-05-22:
Yes, this bugfix has been integrated into both the v1.1 and the master branches. Closing.

closed 2017-05-22 by tswilliams