ipbus/ipbus-firmware #33: Coupling of slaves to the bus

tswilliams, 2018-01-10:
Question from Daniel Montesinos, @Dahernan01 ; transferring here from CACTUS TRAC.

The steps that are described in the IPBus_firmware_notes.pdf to add slaves to the IPBus are not easy to understand for me.

How I can modify the address decoder table without hand edit the ipbus_decode_ipbus_example.vhd?

The I2C and SPI opencores can be slaves of the IPBus firmware?

Sorry about this novice incidence, I read all the documentation (very well explained) but I'm missing something.

Thank you.

dmnewbold, 2018-07-19:
I didn't see this one - is it still open?

Dahernan01, 2018-07-20:
No! You can close it.

dmnewbold, 2018-07-20:
OK - did you solve your issue? If not, please get in touch.

closed 2018-07-20 by dmnewbold

Dahernan01, 2018-07-20:
Yes, everything it's okay.

Thank you.