ipbus/ipbus-firmware #34: STAT reg

mkrivda, 2018-01-24:
We have a universal board which has several firmware types, so we need to have read-only register with FW_TYPE at defined address which is the same for all types of firmware.
We have used STAT reg inside CTRL type reg.
If we add more CTRL reg then address for STAT reg changes as they are created after CTRL regs.
Is it possible to modify firmware in such way that STAT registers are created first ?

tswilliams, 2018-01-24:
Hi,

I assume that you are referring to the `ipbus_ctrlreg_v` entity that's defined within `ipbus_ctrlreg_v.vhd`. If that's not the case, please let me know

I do not believe that it would be possible at this stage to change the order of the STAT & CTRL register within `ipbus_ctrlreg_v`, as that would be a backward-incompatible change that would affect many other people.

Instead, given the fixed-address constraint on the firmware version/type register(s) in your case, I think that a more feasible solution would be for you to declare the CTRL and STAT register blocks as two separate components in place of the single `ipbus_ctrlreg_v` entity that you're currently instantiating. 

Cheers,
Tom

dmnewbold, 2018-02-24:
I have added a new generic SWAP_ORDER to the ctrlreg_v block and its friend syncreg_v. Added to the dev branch for this issue.

dmnewbold, 2018-02-24:
I've also used this branch to fix a recently-discovered race condition in ipbus_syncreg_v. This caused a lock-up condition if a block of syncreg_v stat registers were read too quickly, due to a bug in the read handshake logic.

alessandrothea, 2018-03-01:
Any more work required on this or shall the branch merged back into the master?

dmnewbold, 2018-03-01:
Pull request made.

dmnewbold, 2018-03-01:
I think it looks good, I'll make a pull request?

> Alessandro Thea <mailto:notifications@github.com>
> 1 March 2018 at 15:42
>
> Any more work required on this or shall the branch merged back into 
> the master?

closed 2018-06-21 by tswilliams