ipbus/ipbus-firmware #52: KCU105 example communication problem

kadamidis, 2018-07-04:
Hello,

After implementing the KC705 IPbus basex example I am trying the one for the KCU105 board via SFP Ethernet Interface. The problem is that I take no answer back when trying to talk to the board and no ping at all. I am using the v1.2 firmware and just modified the MAC and IP address to ones we used in the past for one other board at this PC. 
Moreover, I use a Finisar FCLF8521P2BTL SFP and am not sure if it is compatible with the KCU. Can you purpose me any other that is working for sure?
Thanks in advance
Kosmas

tswilliams, 2018-07-04:
Hi,

The quickest way to debug the problem may be to use the Vivado Integrated Logic Analyser Chipscope to monitor the signals out of the ethernet block. If packets are arriving at that level, then it could be an ipbus issue. If no packets arrive,  then it could either be a problem in the communication  between the SFP and the switch, or between the FPGA and the SFP.

Also, at the same time it may be worthwhile checking that various configuration settings on the firmware side, such as auto-negotiation, have consistent values with the PHY in that SFP.

Cheers,
Tom

kadamidis, 2018-07-05:
Hello

Thank you for the reply.
I will try what you purposed me. Any thought on the SFP module I use?

Kosmas

dmnewbold, 2018-07-05:
It might be worth trying with an optical SFP, I suspect the problem is that the MAC core is not autonegotiating with the PHY in your SFP.

kadamidis, 2018-07-06:
Well finally it worked just fine. The problem was that I didn't program the input clock at 156.25 MHz. Thank for your immediate replies.

tswilliams, 2018-07-06:
Glad to hear that it's now working fine. Closing the ticket for now; please re-open if you have further problems.

closed 2018-07-06 by tswilliams

dmnewbold, 2018-07-06:
Aha! Well spotted - apologies this is a 'common problem' with KCU105 also. We ought to start making a list of these.