ipbus/ipbus-firmware #59: error message for ipbus_dpram in VIVADO 2017.4

mkrivda, 2018-07-24:
I am using ipbus fw 1.2.
I have upgraded VIVADO 2016.2. to VIVADO 2017.4 and I am getting error message for ipbus_dpram.

[image]

mkrivda, 2018-07-24:
[image]

dmnewbold, 2018-07-26:
Man that is the nicest formatted bug report ever. If only all bug reports were like this 8-)

dmnewbold, 2018-07-26:
Fixing now.

tswilliams, 2018-09-26:
The fix for this bug was just merged into the `master` branch. Closing the ticket now. @mkrivda , let us know if you still see any related problems.

Thanks,
Tom

closed 2018-09-26 by tswilliams