ipbus/ipbus-firmware #99: Compilation error in ipbus_pipeline.vhd

m-kru, 2019-03-10:
`/ipbus-firmware/components/ipbus_core/firmware/hdl/ipbus_pipeline.vhd:90:32: no declaration for "ipbus_in"`

Looks like it should be `m_ipb_in` instead of `ipbus_in` ? Or maybe `s_ipb_out.ipb_strobe` is also wrong on the left side of the signal assignment?

tswilliams, 2019-03-12:
Hi,

Thanks for reporting this. I've opened a pull request containing a potential fix - changing `ipbus_in.ipb_strobe and not got_ack` to `m_ipb_in.ipb_strobe and not got_resp` - but still need to confirm this.

Cheers,
Tom

tswilliams, 2019-08-15:
Hi,

This fix has now been merged into the `master` branch and will be included in the next tag of the repo. (Sorry for the delay in merging - it slipped off my 'to do' list.)
 
Cheers,
Tom

closed 2019-08-15 by tswilliams