ipbus/ipbus-firmware #115: Valid MAC addresses

moetie, 2019-06-07:
We try to use the IPBus firmware with an ARTIX7 FPGA and enabled RDNA feature.

To get a unique MAC address we using the DNA_PORT information of the FPGA. After trying some different FPGA it seems that the firmware does not work with arbitrary MAC addresses. This is also mentioned in the example projects:

mac_addr <= X"020ddba1151" & dip_sw; -- Careful here, arbitrary addresses do not always work

I can confirm that issue.

ad:2e:50:21:08:54  not working
14:49:4d:80:88:54  workin
4d:0e:50:fb:88:54  not working 

At the moment we using the USR_ACCESS register to alternate the bottom byte but its a lot of trying an error to get a working combination. 

Is it known which MAC scheme works and which not?

tswilliams, 2019-06-13:
Hi,

The IPbus firmware itself as an individual network device can be configured to have any possible MAC address; whether or not any particular MAC address will work in practice within any given network partly depends on the configuration of the other network devices (e.g. switches and computer NICs) on that network.

The significance of certain address ranges is summarised at https://en.wikipedia.org/wiki/MAC_address#Address_details . In particular, the two MAC addresses that you reported as as not working in your network are multicast addresses; if you use only unicast addresses (such as `14:49:4d:80:88:54` that you mentioned) you should have more consistent success.

I hope that helps.
Cheers,
Tom

tswilliams, 2020-01-16:
Hi,

I'm closing this ticket since there have not been any comments for several months. If you still have questions on this topic, please feel free reopen this ticket.

Cheers,
Tom

closed 2020-01-16 by tswilliams