ipbus/ipbus-firmware #122: [Synth 8-97] array index 3 out of range [ipbus_ctrs_ported.vhd:84]

tpsubedi, 2019-08-08:
When I use ipbus_ctrs_ported with N_CTRS = 4, Vivado gives error of "array index out of range" on line 84 of ipbus_ctrs_ported.vhd. In the code "stb(N_CTRS * CTR_WDS - 1) = '1' " is obviously out of range for N_CTRS = 4 and CTR_WDS = 1 provided stb is defined std_logic_vector(0 downto 0); in line 76. 
How do I fix this problem? Thanks

tswilliams, 2019-08-17:
Hi,

I've fixed this bug on branch `bug/122-counter-slaves`. Can you please try checking out that branch, and let me know if you still see any other problems?

Cheers,
Tom

tpsubedi, 2019-08-20:
Hi Tom,
Now it is working. Thanks

closed 2019-08-23 by tswilliams

tswilliams, 2019-08-23:
Hi @tpsubedi ,

Thanks for confirming that the slave is working fine following the bugfix. I've now merged the `bug/122-counter-slaves` branch into the `master`, and created a `v1.5` tag [1]. So I recommend using the `v1.5` tag for future FW builds.

Cheers,
Tom

[1] http://ipbus.web.cern.ch/ipbus/post/firmware_release_1_5/

commit d23eabac (2019-08-17): Fix logic error in ported counter slave; fixes #122
commit e3be8a05 (2019-08-17): Fix build errors (signal width mismatches and multiple drivers) in ported counter slave; fixes #122
commit 05a40bb5 (2019-08-18): Ported counter slave: Fix build errors (signal width mismatches and multiple drivers); fixes #122
commit 7db82dcd (2019-08-18): Ported counter slave: Fix build errors (signal width mismatches and multiple drivers); fixes #122