ipbus/ipbus-firmware #126: Sideband example

GustavSVJ, 2019-09-24:
Dear Developers,

I'm currently using the IPBus in a project. At the moment I'm using a small Python script to initialize the control registers with the correct initial values at boot. When the project is deployed the initial values will have to be read from an EEPROM on the board instead. 
I've considered different ways of doing this and it seems the best way is to add an Out-of-band interface with a connection to the EEPROM. Is that a good solution or is there a better way?

Kind regards,
Gustav

tswilliams, 2019-09-30:
Hi,

> At the moment I'm using a small Python script to initialize the control registers with the correct initial values at boot.

Do these registers have to be set before the IPbus device is accessible over the network? Similarly, are the values read from the EEPROM used to set the board's IP and/or MAC address?

Also, the best method can depend on the details of the connection to the EEPROM: Is it controlled over an I2C bus, or some other interface?

Cheers,
Tom

GustavSVJ, 2019-09-30:
Dear Tom,

The values won't have to be loaded before there is access to the network. The idea is to have a "standard" configuration that can be loaded without having to send all the register values via the network. The EEPROM used will be the same as for the IP and MAC and will be connected via an I2C bus. 

Kind regards,
Gustav

tswilliams, 2019-10-01:
Hi,

If the EEPROM, or other components connected the I2C bus, need to be readable via IPbus as well, then I would recommend adding an out-of-band interface to `ipbus_ctrl` (i.e. using the `oob_in` and `oob_out` ports with the `N_OOB` generic set to 1), and adding an IPbus slave that is a master of the I2C bus. Via a state machine (or on-board microcontroller) that drives the out-of-band interface, on boot you could issue the correct sequence of IPbus transactions to read the contents of the EEPROM, and then set the values of the `ipbus_ctrl` `mac_addr` and `ip_addr` input ports accordingly. 

Otherwise if the I2C components don't need to accessible via IPbus, then you could have a state machine or microcontroller read the EEPROM's value on boot (without issuing IPbus transactions), and then drive the `ipbus_ctrl` `mac_addr` and `ip_addr` input port values accordingly.

Cheers,
Tom

tswilliams, 2019-10-01:
(oops, didn't mean to close)

GustavSVJ, 2019-10-02:
Dear Tom,

Thank you very much for your answer. I've been looking for a bit of documentation on the `oob` interface. I've only found some documentation stating it belongs to the `ipb_clk` domain. 
Does there exist any further documentation? If not could you give a brief description of how the `ipbus_trans_in` interface should be used? 

Kind regards,
Gustav

tswilliams, 2019-10-07:
Hi,

The `oob_` ports can be used to send packets of IPbus transaction requests to the transactor, and receive the responses. The record types for these ports are defined in `components/ipbus_core/firmware/hdl/ipbus_trans_decl.vhd`. The `oob_` ports are both arrays; the elements of the `oob_in` port are of type `ipbus_trans_in`, and the elements of the `oob_out` port are of type `ipbus_trans_out`. If you only have one out-of-band transactor master, then you can set `N_OOB` to 1, and each of those ports will be arrays of length 1.

Packets of IPbus 2.0 transaction requests [1] should be assembled in a RAM, with 32-bit data interfaces. The RAM's read data interface should be connected to the `oob_in` port's `rdata` element, and the RAM's read address port connected to the `oob_out` `raddr` member. The write data interface for another RAM, that will store the reply packet, should be connected to the `oob_out` port's `wdata` element, that RAM's write enable port connected to the `oob_out` `we` member, and that RAM's write address port connected to the `oob_out` `waddr` member.

When a request packet is ready, the `oob_in` `pkt_rdy` signal should be raised high for one clock cycle. Over several clock cycles, the transactor will then parse the request packet (by changing the value of `raddr`), and write the reply packet (by changing the values of `we`, `waddr` and `wdata`). Once the entire request packet has been processed, and the entire reply packet has been created, `pkt_done` will be raised high for one clock cycle. The data from any read transactions can then be extracted by the OOB master logic.

Does that help? Let me know if anything is still unclear.

Cheers,
Tom

[1] See link to IPbus 2.0 specification on http://ipbus.web.cern.ch/ipbus/doc/user/html/FAQ.html#what-does-uhal-stand-for

tswilliams, 2020-01-16:
Hi,

I'm closing this ticket since there have not been any comments for a few months. If you still have questions on this topic, please feel free reopen this ticket.

Cheers,
Tom

closed 2020-01-16 by tswilliams