ipbus/ipbus-firmware #127: Typo-bug in clocks*.vhd

jhegeman, 2019-10-02:
Hi,

It looks as if the assignments of the clkin1_period generic in severeal of the clocks*.vhd files in ipbus_util are not completely correct.

For example: https://github.com/ipbus/ipbus-firmware/blob/master/components/ipbus_util/firmware/hdl/clocks/clocks_usp_serdes.vhd#L104

The statement currently reads:
`clkin1_period => CLK_VCO_FREQ / CLK_FR_FREQ`
which is numerically correct, but not because this has anything to with CLK_VCO_FREQ.
The correct statement should read:
`clkin1_period => 1000.0 / CLK_FR_FREQ`
where the factor 1000.0 converts the period value to nanoseconds.

At first glance this same issue is present in the following files:
- clocks_7s_extphy.vhd
- clocks_7s_extphy_se.vhd
- clocks_7s_serdes.vhd
- clocks_us_serdes.vhd
- clocks_usp_serdes.vhd

Cheers,
Jeroen

tswilliams, 2019-10-03:
Hi,

Good point! I've pushed the fix to branch `bug/127-clocks-period-typo`, will merge that branch in ASAP, and create a new release next week.

Cheers,
Tom

closed 2019-10-07 by tswilliams

tswilliams, 2019-10-22:
Note: This fix has been included in the `v1.6` tag that was created earlier today.

commit df5630f6 (2019-10-07): Fix typo in setting MMCM 'clkin1_period' generic; fixes #127