ipbus/ipbus-firmware #136: Missing vhd file in ipbus_transport_axi dependency file

jhegeman, 2019-12-11:
Hi,

The dep file `components/ipbus_transport_axi/firmware/cfg/ipbus_transport_axi.dep` requires the test-bench file `ipbus_transport_ram_if_simtb.vhd`. It does not, however, require the `ipbus_example` VHDL file, which is used by that test-bench.

Cheers,
Jeroen

tswilliams, 2019-12-11:
Hi,

Good point! Since most people using `ipbus_transport_axi` will not need the sim testbench, I'd prefer instead splitting out `src ipbus_transport_ram_if_simtb.vhd` into a new dep file, `ipbus_transport_axi_simtb.dep`, and including `ipbus_example.vhd` from that dep file.

Cheers,
Tom

jhegeman, 2019-12-11:
Crap ;-) Let me go an cancel my pull request then, and make a new one.

Jeroen

jhegeman, 2019-12-11:
Ok, let's try this one then: https://github.com/ipbus/ipbus-firmware/pull/138.

jhegeman, 2019-12-15:
Ok. This was handled in pull request #138. All happy now.

closed 2019-12-15 by jhegeman