ipbus/ipbus-firmware #139: Genesys2

barryoakgreen, 2019-12-11:
I would like to use IPBus on a Digilent Genesys2 evaluation board. It uses the same FPGA as the KC705 but would require an RGMII interface to the PHY. I can see that each of these is used by different example designs, so suspect it can't be too complicated to customise a project. Is such customisation documented or can you suggest which files I should look at to create a suitable project. I have already successfully created and built various example designs using ipbb.
Thanks
Barry Green

tswilliams, 2019-12-12:
There's no documentation on the website for this level of customisation of the example designs, but I think you should be able to customise the design for your eval board as follows:
 1. Take the KC705 GMII example design as a starting point (board-specific files for that design can be found under `boards/kc705/gmii` and `boards/kc705/common`)
 2. Replace the `eth_7s_gmii` entity in `components/kc705/gmii/firmware/hdl/kc705_gmii_infra.vhd` with `eth_7s_rgmii` (source file: `components/ipbus_eth/firmware/hdl/eth_7s_rgmii.vhd`)
 3. Update the `gmii_` ports in `kc705_gmii_infra.vhd` and `top_kc705_gmii.vhd` to match the `rgmii_` ports from corresponding files in the Enclustra AX3 PM3 design (i.e. `enclustra_ax3_pm3_infra.vhd` and `top_enclustra_ax3_pm3.vhd` in `boards/enclustra_ax3_pm3/synth/firmware/hdl`)
    * `kc705_gmii_infra.vhd`: lines 54 to 61 and lines 125 to 132
    * `top_kc705_gmii.vhd`: lines 45 to 52 and lines 82 to 89
 4. Update the pin constraints in `components/kc705/gmii/firmware/ucf/kc705_gmii.tcl` accordingly.

That said, both of these example designs were written by @dmnewbold , so he'll correct me if there's a flaw in this plan.

barryoakgreen, 2019-12-12:
thanks Tom, that's really helpful
Barry

dmnewbold, 2019-12-12:
Tom's recipe looks OK to me. One other point is that the RGMII clock constraints for clock-data phase are 'fiddly' and sometimes require adjustment depending on board design; you can find details in the Xilinx MAC core documentation.