ipbus/ipbus-firmware #141: Cleanup suggestion: rename syncreg_v.dep to ipbus_syncreg_v.dep

jhegeman, 2019-12-15:
Hi,

In my opinion it would be nice(r) if the dep files had the same base name as the corresponding VHDL files. This is already the case for most of the existing dep files for the IPBus slaves, except for the one corresponding to ipbus_syncreg_v.vhdl, which is called simply syncreg_v.dep.

NOTE: This dep file is already included in several places, of course. In order to make the transition smoother (?) one could rename the file and add a symlink with the original name, maybe?

Cheers,
Jeroen

jhegeman, 2019-12-15:
Maybe, at the same time, one could rename the files syncreg_{r,w}.vhd to ipbus_syncreg_{r,w}.vhd.

alessandrothea, 2019-12-15:
Well spotted, thanks!
The dep file should (and will eventually) be renamed to match the dep file. The only slight hiccup is that this change will break almost every existing ipbus-based design, but such is life.

Regarding `syncreg_{r,w}.vhd`, not necessarily. Those 2 entities are not ipbus specific and that is why they don't have the `ipbus_` prefix at the moment. This might change in the future, as we're discussing how to handle these cases

commit fa90ccf7 (2019-12-18): syncreg slave: Rename depfile to match entity name. Fixes #141
closed 2019-12-19 by tswilliams