ipbus/ipbus-firmware #144: Suggestion: add basic US and USP sysmon interfaces in ipbus_util

jhegeman, 2019-12-16:
Having basic interfaces for the UltraScale(+) SYSMON IPs can be quite convenient, in my opinion. These can serve as quick-check utilities, or as a starting point for a more in-depth implementation.

jhegeman, 2019-12-16:
See pull request #145.

NOTE: The USP versions have been tested, but the US versions remain creative copy-paste since I don't have an US FPGA to test with.

alessandrothea, 2019-12-17:
Hi Jeroen,
Thanks for the ticket/PR. Very interesting addition!
In order to better evaluate the PR code it would be good to have 

- An implementable example for US+ and US boards using the existing example designs (@tswilliams and I need to find a place where to put such examples). 
- In particular demonstrating the differences between external and internal references
- A minimal python script that translates hardware values to physical quantities
- We are trying to provide uniform support (to an extent) for US, USP and Series 7, for which SYSMON was not there, but XADC is. How hard would it be to provide something similar to the SYSMON ipbus interfaces for XADC?

@tswilliams anything else that comes to your mind?

jhegeman, 2019-12-18:
Ok, I had a look at a few boards that I can get my hands on.

- I can definitely integrate this into the VCU118 example designs. This exercises the USP code with internal reference.
- I can also add a little demo script that shows how this works. @alessandrothea: You will have to tell me where this script should go.
- I can probably implement the same thing (called sysmon at the ipbus level for consistency) based on the XADC primitive for the 7-series, and integrate this into the kc705 example designs. With the same demo script, presumably. On this board one can choose (with a jumper) between using the internal or an external voltage reference.

tswilliams, 2019-12-18:
Hi,

So far, I've only thought of one minor thing beyond what Alessandro mentioned above: For test purposes - both (semi-)automated test suites and as a reassuring sanity check in interactive tests - it would be nice if we could expose in the address table some sysmon register for which we know what the exact value should be (e.g. a register with a fixed value for a particular device). However, after looking through for UG580, I haven't found any enticing candidates for such a register yet.

Cheers,
Tom

jhegeman, 2019-12-19:
Myeah... The best you can probably do, as far as I can see, is read the voltage measurements (and convert them to real values) and compare those to what you know they should be. But also these differ from family to family, so I'd say this would be closer to testing the hardware is powered up correctly than to checking the firmware.

jhegeman, 2019-12-19:
Ok, so please have a look at https://github.com/ipbus/ipbus-firmware/pull/145 again. I have added (tested!) example SysMon designs for the KC705 and the VCU118, with little Python demo scripts and even a bit of documentation.

The easiest way to check out what this contains is by having a look at the documentation in my fork:
https://github.com/jhegeman/ipbus-firmware/blob/ipbus_issue144/projects/example/scripts/README

jhegeman, 2020-01-15:
This was addressed in PR #145.

closed 2020-01-15 by jhegeman