ipbus/ipbus-firmware #176: Suppressing Vivado 'critical warning' on ipbus_ported_dpram?

jhegeman, 2020-06-24:
Dear IPBus experts,

The `ipbus_ported_dpram` infers a block RAM (at least in Vivado), but then Vivado also complains about
```
TIMING #6 Critical Warning The clock pin sys_main/flash_interface/buffer_ram/ram_reg_bram_15/CLKBWRCLK is not reached by a timing clock 
```
(and a whole lot more of those).

If I understand correctly, this is because the `rclk` port of `ipbus_ported_dpram` ends up connected to the block RAM clock pins, whereas is it really an acces strobe. Is there a way to explain this to Vivado (in order to get rid of the 'critical warning')?

Best regards,
Jeroen

alessandrothea, 2020-06-25:
Hi Jeroen,

`rclk` is the read clock signal, so I'm not sure what you mean with access strobe.
If you refer to a `read_enable` signal, there's no such thing in `ipbus_ported_dpram`.
The output value `q` always corresponds to the content of the ram at `addr`.

Alessandro

jhegeman, 2020-06-25:
Hi Alessandro,

True: the `rclk` is the RAM clock. I was a bit hasty when writing earlier, and possibly confused by what Vivado is showing me. It appears that Vivado interprets my use in a way that leads to an implementation with one of the block RAM clocks connected to a strobed fabric resource. Maybe I'm using this in an improper way. I will have another look.

Cheers,
Jeroen

jhegeman, 2020-06-25:
Ok, after some digging this indeed turned out to be my bad. I ended up with a gated clock because I did not pay attention in my FSM...

Sorry for the noise.

Jeroen

closed 2020-06-25 by jhegeman