ipbus/ipbus-firmware #177: zcu102_basex project ...  which version of vivado was this originally built with?

kjazatlas, 2020-06-25:
hi,
i have been trying to build zcu102_basex project following the instructions here
https://ipbus.web.cern.ch/ipbus/doc/user/html/firmware/exampleDesigns.html
i tried both with vivado 2018.3 and 2015.4.  in the former all the steps completed but the
project did not build. in part because the autoupdate of the ethernet ip was not successful.
in the latter, the project does not make at the make-project step as i get the error xczu9eg-ffvb1156-2-e part not found.
my question is, is there a version of vivado that would build this out of the box?  which version was this originally tested with? thanks in advance for your help and suggestions.

tswilliams, 2020-06-25:
Hi,

The example designs for the last release were built using Vivado 2018.3 and 2019.1. In particular, the specific version of 2018.3 used in current automated builds (as printed when starting vivado) is:

> ****** Vivado v2018.3.1 (64-bit)
>   **** SW Build 2489853 on Tue Mar 26 04:18:30 MDT 2019
>   **** IP Build 2486929 on Tue Mar 26 06:44:21 MDT 2019

> i tried both with vivado 2018.3 and 2015.4. in the former all the steps completed but the
project did not build. in part because the autoupdate of the ethernet ip was not successful.

Can you clarify what you mean by "the project did not build"? E.g. Did errors occur during implementation/synthesis? Was a bitfile produced?

Cheers,
Tom

kjazatlas, 2020-06-25:
hi,
a.  thank you for responding so quickly!!  unfortunately i am going to be offline for 2 days and i overwrite the 2018.3
version with 2015.4.  but what i had from my first try is below.  if that is not sufficient, i can reproduce, but probably not until sun.
[image]
[image]
[image]
let me know if you can see these.  thanks..kj

alessandrothea, 2020-06-26:
Hi, have you ruled out a licence problem?
Those are ip cores that require specific licences, if I'm not mistaken.

kjazatlas, 2020-06-29:
i am still working on this.  trying to resolve possible license issues that are preventing the building of the bitfile.

kjazatlas, 2020-07-07:
hi,
a.  thanks again for your help and suggestion.
b.  yes, there was an odd license issue that we resolved.  and also we had to use only broadcom sfp module.  anyway, the example worked out of the box as advertised.

closed 2020-07-07 by kjazatlas