ipbus/ipbus-firmware #207: DHCP support

mwensing, 2022-01-24:
Dear IPbus-team,

we are using the IPbus protocol in one of our FPGAs and during a review the RARP-method for IP-address assignment as been commented as obsolete (which is kind of true). I have seen that you are working on a DHCP-version in a feature branch. Can you elaborate on the timescale of DHCP being available and the current status? Could we already make use of this new branch or is it too experimental?

Best Regards
Marius Wensing

dpcsankey, 2022-01-26:
RARP always was obsolete but it was beautifully simple!

The feature branch is up to date with master and is currently undergoing final tests, having been successfully tested against centos7 dhcpd and Windows Server 2016.

Pending tests prior to release are testing against the CERN IT IPAM infrastructure, but these tests are lower priority than Run 3 commissioning...

The implementation is not a complete DHCP implementation, rather it is sufficient for initial IP address assignment, effectively mimicking the BOOTP address assignment scheme.

The most important requirement is that it assumes that the DHCP server offers it a lease that is permanent (as per BOOTP) - it does not implement the subsequent DHCPREQUEST negotiation on lease expiry that a complete DHCP implementation would have.

The steps that it implements are:

It issues DHCPDISCOVER at random intervals until it receives a valid DHCPOFFER.

It then immediately assumes that the IP address offered is valid (so just like RARP and BOOTP).

To better behave in a DHCP environment it then issues a broadcast DHCPREQUEST for this address (so better corresponding to the DHCP negotiation) but does not wait for the DHCPACK confirmation.

Having received an IP address it assumes that this will remain valid indefinitely, so the DHCP server must have a permanent reservation for this and not have assigned it from the DHCP pool.

Kind regards,

Dave

> Dear IPbus-team,
>
> we are using the IPbus protocol in one of our FPGAs and during a review the RARP-method for IP-address assignment as been commented as obsolete (which is kind of true). I have seen [...]

This email and any attachments are intended solely for the use of the named recipients. If you are not the intended recipient you must not use, disclose, copy or distribute this email or any of its attachments and should notify the sender immediately and delete this email from your system. UK Research and Innovation (UKRI) has taken every reasonable precaution to minimise risk of this email or any attachments containing viruses or malware but the recipient should carry out its own virus and malware checks before opening the attachments. UKRI does not accept any liability for any losses or damages which the recipient may sustain due to presence of any viruses.

mwensing, 2022-01-28:
Thanks for your nice explanation. I will try out your feature branch and test it in my setup and close this issue, since all of my questions have been answered.

closed 2022-01-28 by mwensing

dpcsankey, 2022-01-28:
Just to add that the way to implement this

Ipbus_ctrl has a new GENERIC DHCP_RARP, which with the default value ‘0’ implements the RARP logic, when set to ‘1’ implements DHCP:
+-- Switch between using DHCP or RARP as the protocol for external IP address management
+-- '0' => RARP, '1' => DHCP
+               DHCP_RARP: std_logic := '0';

There is then a renamed port ipam_select which selects whether a static IP address is assumed when set to the default value ‘0' or one is requested from whichever of RARP or DHCP is enabled when set to ‘1':
-               RARP_select: in std_logic := '0';
+               ipam_select: in std_logic := '0';

And to reiterate that the DHCP server must have a permanent reservation for the board as the firmware does not implement lease renewal - if it is given an IP address from the pool the server will not know that it is still in use and may reassign the IP address to a different machine, which will result in carnage.

I’d appreciate any news of how you get on with it and details of what DHCP server you use.

Kind regards,

Dave

> 
> 
> Closed #207.

This email and any attachments are intended solely for the use of the named recipients. If you are not the intended recipient you must not use, disclose, copy or distribute this email or any of its attachments and should notify the sender immediately and delete this email from your system. UK Research and Innovation (UKRI) has taken every reasonable precaution to minimise risk of this email or any attachments containing viruses or malware but the recipient should carry out its own virus and malware checks before opening the attachments. UKRI does not accept any liability for any losses or damages which the recipient may sustain due to presence of any viruses.