ipbus/ipbus-firmware #221: "ipbb ipbus gendecoders" does not work

abunickabhi, 2023-07-07:
While setting up a firmware project, I follow the instructions listed [https://serenity.web.cern.ch/serenity/emp-fwk/walkthroughs/walkthrough1.html#walkthrough1](url) here.

I am unable to `ipbb ipbus gendecoders` as it does not recognize ipbus as a valid keyword.

How do I execute this command successfully? What does this command do exactly?

tswilliams, 2023-07-07:
Hi,

Can you paste the output of the command here?

Thanks,
Tom

abunickabhi, 2023-07-10:
Hi,
Here is the output when I execute the command in ipbb environment.

```
(ipbb) (base) abhi@abhi:~/Desktop/CMS_Trigger_TIFR/ipbb/algo-work/proj/my_algo$ ipbb ipbus gendecoders
Usage: ipbb [OPTIONS] COMMAND [ARGS]...
Try 'ipbb -h' for help.

Error: No such command 'ipbus'.
```

abunickabhi, 2023-07-17:
What could be the issue for not recognising the `ipbus` keyword @tswilliams ?

tswilliams, 2023-07-17:
I'm not completely sure what the cause of this could be. What operating system are you using? Could you try skipping that command, and running the `ipbb vivado` commands listed after that?

abunickabhi, 2023-07-17:
I am using Ubuntu 20.04.

Skipping the command is working as I am able to create a bitstream using `ipbb vivado`.

Is the `ipbus gendecoders` command optional while making a bitstream project for Serenity?

tswilliams, 2023-07-17:
Yeah, as long as you don't need to add address decoding in the payload (and this is the case for most payloads right now), you don't need to run the gendecoders command

abunickabhi, 2023-07-17:
Got it thanks!

closed 2023-07-17 by abunickabhi