ipbus/ipbus-firmware #243: ZCU102 example design license issue with  temac_gbe IP

joschambach, 2024-08-02:
I am trying to compile your example design for the ZCU102 eval board, but get an error message during the bitstream generation:

[Common 17-69] Command failed: This design contains one or more cells for which bitstream generation is not permitted:
infra/eth/mac/U0/temac_gbe_v9_0_core (<encrypted cellview>)
If a new IP Core license was added, in order for the new license to be picked up, the current netlist needs to be updated by resetting and re-generating the IP output products before bitstream generation.

Is a license required to use this example design?

andrea-celentano, 2024-10-25:
Dear colleagues,
we are facing the same issue for a Kintex-7 based hardware for a CERN project (NA64 experiment).
We are compiling at CERN using  for `$XILINXD_LICENSE_FILE` the following settings:
 
`2112@lxlicen01,2112@lxlicen02,2112@lxlicen03,2112@licenxilinx`

May I ask you if a proper license server at CERN exists that we are authorized to use to generate the bitstream for our project?

Thanks,
Bests 
Andrea Celentano

jhegeman, 2024-10-25:
Hi Andrea,

As far as I know, licenxilinx, _is_ the proper CERN Xilinx license server.
I use the following configuration:
```export XILINXD_LICENSE_FILE="2112@licenxilinx"```
and 'things work for me'.

If you are looking for information on a specific license (at CERN), the right person to ask Christos Gentsos.

Best regards,
Jeroen

andrea-celentano, 2024-10-25:
Hi Jeroen,
thanks! Indeed, after setting the `XILINXD_LICENSE_FILE` to `""2112@licenxilinx"` only, the project compiles properly. I am curious why it did not work before with the four servers setting... Anyhow, thanks for your suggestion!

closed 2024-10-28 by tswilliams