ipbus/ipbus-firmware #252: OS vs Vivado

AleCamplani, 2025-08-25:
Hi :) I would like to start using IPbus for a KCU105.
I have been looking at the Vivado versions that are supported (up to 2022.2), but they seems to collide with Alma9 OS.

What would be your recommendation, as I need to work on Almalinux9? 
Do you have some recent beta-testing with a Vivado supported this OS? Or can we work together on that, for the KCU105?

Thanks in advance for the help!
Alessandra :)

jhegeman, 2025-08-26:
Hi Alessandra,

I'm not sure where you found the 2022.2 limit, but I'm using that Vivado version on Alma 9 without problems.
I'm also sure that there people using IPBus with Vivado 2023 and possibly even newer.)

(The IPBus users' guide at https://ipbus.web.cern.ch/doc/user/html/firmware/index.html#supported-devices-and-toolsets says it supports 'Vivado 2018.3 and 2019.2' but that is definitely out-of-date.)

Cheers,
Jeroen

AleCamplani, 2025-08-26:
Dear Jeroen, 
the 2022.2 version is also from the user guide, for the latest release: https://ipbus.web.cern.ch/doc/user/html/firmware/releases/v1_16.html

So, 2023.2/1 on Alma9 for you is working! That's good to hear :)
I will then try it.

Thanks for your feedback.
Alessandra

dpcsankey, 2025-08-26:
You could also talk to Francesco, we have switched to 2024.2 and 2025.1.

D

tswilliams, 2025-09-08:
I've updated the documentation to correctly reflect the range of tool versions that's currently used.

Cheers,
Tom